claws/BH1750 #7: How can I set I2C adress?

FrankenApps, 2016-07-03:
I use the sensor together with other I2C devices on one arduino and therfore would like to set a unique I2C  adress for the sensor.

FrankenApps, 2016-07-03:
I know that I can change it by setting a new value here: `#define BH1750_I2CADDR 0x23`, but it would be nice if we would be able to change it in the code directly, not in the library...

claws, 2016-07-03:
It should be possible to support this, within the constraints of the device's limitations. The device only supports two addresses; 0x23 and 0x5C. With ADD not connected the address is 0x23 and when ADD is pulled high the address becomes 0x5C.

FrankenApps, 2016-07-06:
Ahh, ok i should have looked up, what this addr pin is for... Thank you very much!

closed 2016-07-06 by FrankenApps