claws/BH1750 #12: Changing the default pins.

PedroD, 2017-02-15:
Hi!

Is it possible to change the default A5 and A4 pins (both in Arduino and ESP8266)?

Thanks!

Great lib!!

claws, 2017-02-18:
It [looks](https://www.arduino.cc/en/Reference/Wire) like the pins for SDA and SCL are defined in the Arduino header wire.h. It does not look like it can be changed there.

I'm not yet sure about the esp8266.

vasco65, 2017-04-05:
In the esp8266 it works ok with other pins, just have to setup i2c before using it with `Wire.begin(sda_pin,scl_pin);`
I'm using in this moment `Wire.begin(0,2)` on a NodeMCU (ESP-12) and connecting sda to pin D3 and scl to pin D4.

Sdsyc, 2017-04-09:
For "**WEMOS D1 mini Pro**" wifi board based on **ESP-8266EX**. It works perfectly with:

//  Initialize WEMOS D1 mini Pro
//  D1  IO,  SCL GPIO5
//  D2  IO,  SDA GPIO4
  **Wire.begin(4,5);** //sda_pin 4 ,scl_pin 5

claws, 2017-04-15:
Question answered, closing.

closed 2017-04-15 by claws