claws/BH1750 #17: Reading light level fails if using the ONETIME modes

gerritv, 2017-08-24:
These modes power down the chip after the level is read. The present code does not wake up the chip when reading the value in those modes.

I added a save of mode and changed the code in readLightLevel:

`// Start transmission to sensor
	Wire.beginTransmission(BH1750_I2CADDR);
	switch (MODE) {

	case BH1750_ONE_TIME_HIGH_RES_MODE:
	case BH1750_ONE_TIME_HIGH_RES_MODE_2:
	case BH1750_ONE_TIME_LOW_RES_MODE:
		__wire_write((uint8_t)MODE);
		break;
	}
	// Read two bytes from sensor
	Wire.requestFrom(BH1750_I2CADDR, 2);`

closed 2017-08-27 by claws

gerritv, 2017-08-27:
Awesome, thank you!!!

coelner, 2017-08-29:
Please add the right delay time for the one-time measurement (datasheet page 7 and page 5 )

claws, 2017-08-31:
@coelner, if you know what to do you are welcome to submit a merge request with the proposed change.

PR #19 (2017-08-27): Fix One-Time use case
This change applies the suggested changes to fix #17. 
It also adds a script to check examples build locally and cleans up some simple typos and grammar.