claws/BH1750 #25: Using 2 sensors from one code

wixa, 2017-12-10:
Hello. Thank you for your library. I have question.
How can i using 2 sensors by one  pair I2C pins? I know that i can setup 2 adress on sensor, by ADD pin, but  how i can read data from both sensors wich have 2 diferent adress from code? How init 2 sensors by code?

HeavensBlade, 2017-12-11:
Try this
```
#include <Wire.h>
#include <BH1750.h>
BH1750 lightMeter1(0x23);
BH1750 lightMeter2(0x5C);
void setup(){
  Serial.begin(9600);
  Wire.begin();
  lightMeter1.begin(BH1750_CONTINUOUS_HIGH_RES_MODE);
  lightMeter2.begin(BH1750_CONTINUOUS_HIGH_RES_MODE);
  Serial.println(F("BH1750 Test"));
}
void loop() {
  uint16_t lux1 = lightMeter1.readLightLevel();
  uint16_t lux2 = lightMeter2.readLightLevel();
  Serial.print("Light1: ");
  Serial.print(lux1);
  Serial.print(" lx. Light2: ");
  Serial.print(lux2);
  Serial.println(" lx.");
  delay(1000);
}
```

claws, 2017-12-29:
Thanks @HeavensBlade for providing useful help to @wixa. I am assuming that this resolves the issue so I'll close it out.

closed 2017-12-29 by claws