claws/BH1750 #26: Improve error handling

claws, 2017-12-29:
The problem highlighted in this issue is pretty unlikely and has never been reported before so is likely very low priority.

When reading a light value there is a possibility that the data read is not the expected 2 bytes. The implementation assumes that 2 bytes are read, even if they are not, and continues progressing through the function which could then use garbage data in the uninitialised ``level`` variable.

Only a happy path is really accommodated for in the implementation and Arduino does not seem to really support error handling (e.g. exceptions). Perhaps we could consider returning a negative number to signal an error - though the existing return value type does not support that. Perhaps in this condition it could return 65535 which is well above the maximum possible lux value (which is raw/1.2 = 54612).

coelner, 2018-01-05:
Maybe there is another flaw: If we use H-Resolution Mode2 we get a 0.5 resolution. Therefore we need a float as return value. And in that case we could use negative numbers

bjung82, 2018-02-26:
I added an else-path at BH1750.cpp:183
with 

Wire.reset();

Without this the I2C bus was held down forever. It took me some hours
because the problem was not easily reproducible.

coelner, 2018-02-28:
@bjung82 explanation @ #41

coelner, 2018-10-25:
Any issue left or should this topic be closed?