claws/BH1750 #35: Define conversion factor as a constant

claws, 2018-01-05:
After reading a raw value read from the sensor it is converted to lux units by dividing by a conversion factor. This value is typically 1.2. For use in this library where no calibration is being performed the factor is effectively a constant. 

The value is currently hard coded in the body of the ``readLightLevel`` function and looks like a magic number.

Perhaps a better alternative (e.g. clean, more maintainable) would be to define it as a constant along with some comments and use the constant. Something like this:

``` c
// Correction factor used to calculate lux. Typical value is 1.2 but can range from 0.96 to 1.44.
// See data sheet (p.2, Measurement Accuracy) for more information.
const float BH1750_CONV_FACTOR = 1.2;
```

and then later when being used:

``` c
// Convert raw value to lux
level /= BH1750_CONV_FACTOR;
```

andrew-from-pretoria, 2018-06-01:
I was disappointed to discover that the readLightLevel function does not return calibrated Lux values. I should have read the device specs better and perhaps elsewhere as well. Anyhow, I proceeded to compare readLightLevel function values to those from a calibrated professional instrument Goldilux GAL 2L Lux meter and here I share some of the results: ( interpret as follows readLightLevel function | Goldilux ) 78 | 178, 53 | 128, 36 | 67, 18 | 28. It appears that the difference is linear.

claws, 2018-10-03:
Closed by #46

closed 2018-10-03 by claws