claws/BH1750 #36: [BH1750] Device is not configured!

dhanushmk, 2018-01-08:
i am getting this in serial monitor
t: 0 lx
BH1750 Test
[BH1750] Device is not configured!

claws, 2018-01-08:
That error gets reported from ``readLightLevel`` if BH1750_MODE attribute is not set. The BH1750_MODE attribute is set in ``configure`` which is called from the ``begin`` function. It is initialised to UNCONFIGURED and is defaulted to CONTINUOUS_HIGH_RES_MODE in the declaration of the the ``begin`` function. 

The typical usage of this library is:
``` c
#include <BH1750.h>

BH1750 lightMeter;

void setup(){
  lightMeter.begin();
}

void loop() {
  uint16_t lux = lightMeter.readLightLevel();
}
```

My best guess would be that perhaps you are not calling ``begin`` to initialise and configure the BH1750 library.

The serial output you pasted does not look like any of the examples. Was this output from an example?
Are you using the current version of the library?
What platform are you running on?
What is the code you are running?

Amy1610, 2018-01-09:
I am having the exact same problem...

Here is my code:

#include <Wire.h>
#include <BH1750.h>

BH1750 lightMeter(0x23);

void setup(){

  Serial.begin(9600);

  Wire.begin();

  lightMeter.begin();
    Serial.println(F("test"));

}

void loop() {

  uint16_t lux = lightMeter.readLightLevel();
  Serial.print("Light: ");
  Serial.print(lux);
  Serial.println(" lx");
  delay(1000);

}

[image]

Amy1610, 2018-01-09:
I downloaded the library yesterday and need to use it for a uni project due soon,

could it be the way I have set up my arduino?:

[image]
[image]
[image]

Amy1610, 2018-01-09:
I just ran the 12C-scanner and it is not detecting the light sensor

Amy1610, 2018-01-09:
I just found that the 12C scanner cannot find the device

claws, 2018-01-10:
It looks like you have wired it up correctly from the pictures. 

The BH1750 library returns a boolean value from ``begin`` to indicate whether the device was configured successfully. You could print that out to check what it says. Modify your ``setup`` function to initialise the lightMeter like this:
```
  if (lightMeter.begin()) {
    Serial.println(F("BH1750 initialised"));
  }
  else {
    Serial.println(F("Error initialising BH1750"));
  }
```

When I run the i2c_scanner you mentioned I see the following output:

```
I2C Scanner
Scanning...
I2C device found at address 0x23  !
done
```

I'm going to guess that perhaps your sensor device is faulty. Do you have access to another to test this out?

I have just run all three of the examples on 2 separate sensor devices and they work as I expect.
When I run the simple example without plugging in the light sensor board I observed the following output:

```
BH1750 Test begin
[BH1750] Device is not configured!
Light: 0 lx
```

You could remove the wire connect ADD to GND. It is not required. The library uses the i2c address of 0x23 by default. I see you were explicitly defining the device address as 0x23 and wiring the ADD line to ground. That is fine and explicit. The same configuration would be achieved by not wiring the ADD line at all and not explicitly passing 0x23 when instantiating the lightMeter - such as in the [simple example](https://github.com/claws/BH1750/blob/master/examples/BH1750test/BH1750test.ino).

Amy1610, 2018-01-10:
I modified my setup function to initialise the lightmeter as you have stated and it returns:
"[BH1750] Device is not configured!
Light: 0 lx" still rather than either printing "BH1750 initialised" or "Error initialising BH1750"

Yeah, the light sensor is probably faulty. I have ordered another one which should arrive Friday. Meanwhile I will have to hope that the code i have written, to include the device in my project, is correct as I have no way of testing it until the sensor arrives :-/

Thanks for your help and the speedy reply! 

Would you be willing to look over my project code to see if it makes sense? It would be really helpful as I am very new to using arduino. Although I understand if you don't want to or do not have time to.

EdrickeAPM, 2018-01-29:
Hi! I'm Working on my BH1750 Light Sensor. I tried running the codes from this forum but there are no readings from my sensor the Serial monitor just displays:

[image]

BTW this is my code:

[image]

Thank you in advance for your response

priema, 2018-02-07:
@EdrickeAPM I have the exact same issue .. have you fixed it yet? :)

coelner, 2018-03-10:
@priema @EdrickeAPM Do you tried another I2C device? Maybe the cable are not in a good shape. 
Change following:
`uint16_t lux = lightMeter.readLightLevel();` --> `uint16_t lux = lightMeter.readLightLevel(true);`

dvbahder, 2018-04-04:
I have the exact same issue .. have you a better fix? thx

coelner, 2018-04-05:
@dvbahder No. Has your sensor worked in the past?

dvbahder, 2018-04-05:
no, I bought the sensor three weeks ago. OneWire DS18B20 works, wire DHT22 works, Display I2C works, as a single sensor BH1750 does not work, I read from a scanner for I2C and want to try it tomorrow. I'll let you know. thx to coelner

dvbahder, 2018-04-06:
The scanner says the address is correct (0x23 GND default), (0x5c high).
PIN SCL SDA was swapped, correct SDA / SCL (D4, D3) ...
is runnig :) thx 
[image]
[image]

Xj98, 2018-06-13:
what is the purpose for adding readLightLevel (true)?

Alacika, 2018-06-20:
The code works: Wemos D1 mini
The code does not work: Wemos D1 Pro and Wemos D1 Lite
I do not know why. I tried everything.

Then I uploaded the ESPeasy firmware and sensor works fine, print lux info in serial console.
I2C scanner sketch says: Found address: 35 (0x23), but it does not work
I tried 0x5C address (ADDR to VCC), but it does not work.
I have 2 Wemos D1 Mini which goes well, and I have 3 other Wemos D1 Mini that do not work anyway.

Sorry for my poor english!

coelner, 2018-06-21:
@Alacika 

1. Do you use the same sensor for each wemos board?
2. Did the ESPeasy firmware work on all your wemos boards, including D1 Pro and D1 lite?
3. Post your sketch

Alacika, 2018-06-21:
1. Yes! (But I tried another BH1750 sensors too, and the problem is the same.)
2. I tested ESPeasy firmware with only D1 Pro
3. Uhh, my sketch is not so professional. :-) [https://gist.github.com/Alacika/2d06d2c96b5caecf99f7ec2eaad32748](url)

coelner, 2018-06-21:
1. At least one sensor is working. Some of your boards not. --> Some boards are broken/not usable.
2. Different pin out?  Wrong pin settings in the arduino toolchain? I don't have those boards, only a guess
3. Did you try the example sketch from this library? And another problem: https://github.com/Seeed-Studio/Grove_BME280/blob/bcf41a39e306a9b435c1140740ff0d7fc1001c45/Seeed_BME280.cpp#L5
remove this.

Alacika, 2018-06-21:
1. Yes.  MyWemos D1 Mini (v2.x) and Wemos D2 Mini (v3.x) work well with my all (4pcs) BH1750 sensors. All other NodeMCU types work well, only my Wemos D1 Pro and Wemos D1 Lite do not work with any BH1750 sensor.
2. The pins are the same: [Wemos versions](https://i.imgur.com/1dxt3kB.jpg) BME280 sensor works on the same I2C bus. Ok, I will mark a comment on "Wire.begin();" line, although the bh1750 error is earlier than executing bme280.init().
3. Yes, I tried your sketches too, unsuccessful. I also tried [Koepel's sketch,](https://gist.github.com/Koepel/b70f81c71a52d8d6d3da86b9fe56d50e) and I remember it worked (not all modes) but tonight I try it again.

Alacika, 2018-06-21:
Some new test logs:

Wemos D1 Pro (ESPEasy)
250 : INIT : Booting version: v2.0-20180316
250 : INIT : Cold Boot
251 : FS : Mounting...
276 : FS : Mount successful, used 75802 bytes of 957314
395 : INIT : Free RAM:22488
395 : INIT : I2C
395 : INIT : SPI not enabled
1288 : WIFI : Connecting SSID attempt #1
5206 : WD : Uptime 0 ConnectFailures 0 FreeMem 19680
5363 : BH1750 Address: 0x23 Mode: 0x99 : Light intensity: 27.50
16026 : BH1750 Address: 0x23 Mode: 0x99 : Light intensity: 28.32
26651 : BH1750 Address: 0x23 Mode: 0x99 : Light intensity: 33.33
37288 : BH1750 Address: 0x23 Mode: 0x99 : Light intensity: 33.33

Wemos D1 Pro (BH1750test.ino example sketch from this library)
[BH1750] ERROR: received NACK on transmit of address
[BH1750] ERROR: received NACK on transmit of data
[BH1750] ERROR: other error
[BH1750] ERROR: undefined error
BH1750 Test begin
[BH1750] Device is not configured!
Light: 0 lx
[BH1750] Device is not configured!
Light: 0 lx
[BH1750] Device is not configured!

coelner, 2018-06-21:
There is a problem with i2c in a generic way. wrong settings in the ide, wrong parameter in the toolchain, It is not a problem of this library. You could use a logic analyzer to research this issue.

quintendewilde, 2018-12-20:
I somewhat managed to fix this by wire.begin(D1, D2) 
I get not error but just blank serial monitor...

coelner, 2018-12-20:
@quintendewilde Some more infos about the board, wiring and sketch?

chengzee, 2020-03-28:
hi, I have the similar issue and try above method but not work for me.
My device is mega2560. 
This is the code I used
`/*

  Example of BH1750 library usage.

  This example initialises the BH1750 object using the default high resolution
  continuous mode and then makes a light level reading every second.

  Connection:

    VCC -> 3V3 or 5V
    GND -> GND
    SCL -> SCL (A5 on Arduino Uno, Leonardo, etc or 21 on Mega and Due, on esp8266 free selectable)
    SDA -> SDA (A4 on Arduino Uno, Leonardo, etc or 20 on Mega and Due, on esp8266 free selectable)
    ADD -> (not connected) or GND

  ADD pin is used to set sensor I2C address. If it has voltage greater or equal to
  0.7VCC voltage (e.g. you've connected it to VCC) the sensor address will be
  0x5C. In other case (if ADD voltage less than 0.7 * VCC) the sensor address will
  be 0x23 (by default).

*/

#include <Wire.h>
#include <BH1750.h>

BH1750 lightMeter;

void setup(){

  Serial.begin(9600);

  // Initialize the I2C bus (BH1750 library doesn't do this automatically)
  Wire.begin();
  // On esp8266 you can select SCL and SDA pins using Wire.begin(D4, D3);

  lightMeter.begin();

  Serial.println(F("BH1750 Test begin"));

}

void loop() {

  float lux = lightMeter.readLightLevel();
  Serial.print("Light: ");
  Serial.print(lux);
  Serial.println(" lx");
  delay(1000);

}`

I only connect a GY-302 (BH1750) sensor.
This is how I connect the wire :
VCC -> 3V3 or 5V (I have tried each of them)
GND -> GND
SCL -> SCL 
SDA -> SDA 
ADD -> not connected

and this is the result 
`BH1750 Test begin
[BH1750] Device is not configured!
Light: -2.00 lx
[BH1750] Device is not configured!
Light: -2.00 lx
[BH1750] Device is not configured!
Light: -2.00 lx
[BH1750] Device is not configured!
Light: -2.00 lx
[BH1750] Device is not configured!
Light: -2.00 lx
`
Could you help me please?

coelner, 2020-03-28:
Which version of this library do you use? Did you verify the wiring with another library or atl east a i2c scan?
you could check the return value from lightMeter.begin() and reconfigure the sensor.

chengzee, 2020-03-29:
Thanks for your answer! 
1. I download the ZIP and use examples/BH1750test. sorry, I don't know which version I used. could you help me find out?
2. It seems like it's my i2c problem. Isn't it default 0x23 without connect ADDR? How to fix it?
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          XX XX XX XX XX XX XX XX XX XX XX XX XX
10: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX
20: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX
30: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX
40: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX
50: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX
60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX
70: XX XX XX XX XX XX XX XX

[BH1750] ERROR: other error
[BH1750] Device is not configured!
Light: -2.00 lx

```
I'm Sorry with so many question and very thanks for your help again. Thanks.

IoTThinks, 2020-03-29:
My esp32 works fine with this library.

Make very SURE that the pins are solidered properly.
If the pins are loosen when the board boots, it will show "Device not configured" forever.

coelner, 2020-03-29:
@chengzee 
1. You use the latest version, it is fine. 
2. You have a general problem with i2c. There should be something at 0x23 or 0x5C. Do you have an alternative sensor? especially one which works? Do you use the pins| 20 (SDA), 21 (SCL)?
3. The examples are not fail save. You could use this for a better understanding:
```
#include <Wire.h>
#include <BH1750.h>

bool BH1750Check = false;
BH1750 lightMeter;

void setup() {

  Serial.begin(9600);

  // Initialize the I2C bus (BH1750 library doesn't do this automatically)
  Wire.begin();
  // On esp8266 you can select SCL and SDA pins using Wire.begin(D4, D3);

  BH1750Check = lightMeter.begin();
  if (BH1750Check) {
    Serial.println(F("BH1750 Test begin"));
  }
  else {
    Serial.println(F("BH1750 Initialization FAILED"));
    while (true) //flow trap
    {}
  }

}

void loop() {

  if (BH1750Check) {
    float lux = lightMeter.readLightLevel();
    Serial.print("Light: ");
    Serial.print(lux);
    Serial.println(" lx");
  }
  delay(1000);

}
```

chengzee, 2020-03-29:
@coelner @IoTThinks Thanks for your answers!!!!! I learn a lot from you guys, thank you.
And it works!!
I found it just like @IoTThinks  said. The sensor need to be very strongly connect. If it is loose, it won't have correct value and i2c would be failed. So I took a looooot of time to solder it (it's my first time...btw, I burned the first one BH1750).
Finally, it works very well.
I'm very appreciate you guys. Thank you!
[image]
[image]
[image]

jafidiesel, 2020-11-15:
In my case it was very weird. I had the same issue but connecting ir to 5v instead of 3.3v do the trick.
I used a arduino nano and a GY-302
The only diference it's that i didn't use ADDR conection since i was using only one sensor.

TaLucGiaHoang, 2021-01-11:
[Solved] After changing new wires the example worked fine. Thanks for your code. It took me hours to figure out

steinbeck, 2021-01-20:
@jafidiesel 
> In my case it was very weird. I had the same issue but connecting ir to 5v instead of 3.3v do the trick.
Oh man, thanks for that. This was the problem also for me when migrating my code from a wemos D1 where things worked fine with the BH1750 powered with 3.3.V to an ESP32. Moving the sensor to 5V on the ESP32 did the trick indeed. What a relief :)

ShSWork, 2021-11-13:
[Solved] I see BH1750advanced.ino that code:

`    Full mode list:

      BH1750_CONTINUOUS_LOW_RES_MODE
      BH1750_CONTINUOUS_HIGH_RES_MODE (default)
      BH1750_CONTINUOUS_HIGH_RES_MODE_2

      BH1750_ONE_TIME_LOW_RES_MODE
      BH1750_ONE_TIME_HIGH_RES_MODE
      BH1750_ONE_TIME_HIGH_RES_MODE_2

  */

  // begin returns a boolean that can be used to detect setup problems.
  if (lightMeter.begin(BH1750::CONTINUOUS_HIGH_RES_MODE)) {
    Serial.println(F("BH1750 Advanced begin"));
  }
  else {
    Serial.println(F("Error initialising BH1750"));
  }
`
I think we have to use MODE in lightMeter.begin() . 

And, in case WEMOS D1 R1 board (Arduino type) I use 
SDA/D4
SCL/D3 

but, I use GPIO pins numbers in the sketch. So:

Wire.begin(4, 5);  

Because at the WEMOS D1 R1 board (Arduino type)
SDA/D4 = GPIO4
SCL/D3 = GPIO5

chess-levin, 2022-02-14:
Hello @claws 

Trying your BH1750 library with an Wemos D1 mini. Using 3.3V for the [BH1750FVI](https://de.aliexpress.com/item/1005001636541540.html?gatewayAdapt=glo2deu&_randl_currency=EUR&_randl_shipto=DE&src=google&src=google&albch=shopping&acnt=494-037-6276&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=12554800262&albag=127990761348&trgt=1284054470089&crea=de1005001636541540&netw=u&device=c&albpg=1284054470089&albpd=de1005001636541540&gclid=EAIaIQobChMIx4Kos8D_9QIVgbt3Ch0ogQAGEAQYAyABEgJwi_D_BwE&gclsrc=aw.ds&aff_fcid=b0ac83152dc648ffb5f9f3d46c49ede7-1644852328281-08689-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=b0ac83152dc648ffb5f9f3d46c49ede7-1644852328281-08689-UneMJZVf&terminal_id=93a3ea88bc5e4dfbb1a05af52fea2ec5) and D1,D2 of the ESP8266 for I2C.

I've added i2cdetect to show, that the BH12750 is detected!

```

/*

Advanced BH1750 library usage example

This example has some comments about advanced usage features.

Connections

  - VCC to 3V3 or 5V
  - GND to GND
  - SCL to SCL (A5 on Arduino Uno, Leonardo, etc or 21 on Mega and Due, on
    esp8266 free selectable)
  - SDA to SDA (A4 on Arduino Uno, Leonardo, etc or 20 on Mega and Due, on
    esp8266 free selectable)
  - ADD to (not connected) or GND

ADD pin is used to set sensor I2C address. If it has voltage greater or equal
to 0.7VCC voltage (e.g. you've connected it to VCC) the sensor address will be
0x5C. In other case (if ADD voltage less than 0.7 * VCC) the sensor address
will be 0x23 (by default).

*/

#include <BH1750.h>
#include <Wire.h>

#include "i2cdetect.h"

/*
  BH1750 can be physically configured to use two I2C addresses:
    - 0x23 (most common) (if ADD pin had < 0.7VCC voltage)
    - 0x5C (if ADD pin had > 0.7VCC voltage)

  Library uses 0x23 address as default, but you can define any other address.
  If you had troubles with default value - try to change it to 0x5C.

*/
BH1750 lightMeter(0x23);

void setup() {

  Serial.begin(115200);

  // Initialize the I2C bus (BH1750 library doesn't do this automatically)
  Wire.begin(D2,D1);
  // On esp8266 you can select SCL and SDA pins using Wire.begin(D4, D3);

  /*

    BH1750 has six different measurement modes. They are divided in two groups;
    continuous and one-time measurements. In continuous mode, sensor
    continuously measures lightness value. In one-time mode the sensor makes
    only one measurement and then goes into Power Down mode.

    Each mode, has three different precisions:

      - Low Resolution Mode - (4 lx precision, 16ms measurement time)
      - High Resolution Mode - (1 lx precision, 120ms measurement time)
      - High Resolution Mode 2 - (0.5 lx precision, 120ms measurement time)

    By default, the library uses Continuous High Resolution Mode, but you can
    set any other mode, by passing it to BH1750.begin() or BH1750.configure()
    functions.

    [!] Remember, if you use One-Time mode, your sensor will go to Power Down
    mode each time, when it completes a measurement and you've read it.

    Full mode list:

      BH1750_CONTINUOUS_LOW_RES_MODE
      BH1750_CONTINUOUS_HIGH_RES_MODE (default)
      BH1750_CONTINUOUS_HIGH_RES_MODE_2

      BH1750_ONE_TIME_LOW_RES_MODE
      BH1750_ONE_TIME_HIGH_RES_MODE
      BH1750_ONE_TIME_HIGH_RES_MODE_2

  */

  Serial.println(F("\n i2cdetect is scanning I2C-Bus: "));
  i2cdetect();
  // begin returns a boolean that can be used to detect setup problems.
  if (lightMeter.begin(BH1750::CONTINUOUS_HIGH_RES_MODE)) {
    Serial.println(F("BH1750 Advanced begin"));
  } else {
    Serial.println(F("Error initialising BH1750"));
  }
}

void loop() {
  if (lightMeter.measurementReady()) {
    float lux = lightMeter.readLightLevel();
    Serial.print("Light: ");
    Serial.print(lux);
    Serial.println(" lx");
  }

  delay(2000);
}
```

This is the output:

```
 i2cdetect is scanning I2C-Bus:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- 23 -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

BH1750 Advanced begin
Light: -1.00 lx
Light: -1.00 lx
Light: -1.00 lx
Light: -1.00 lx
Light: -1.00 lx
```

Is there anyone out there who can publish a working solution? I've already spent hours on this trivial "problem". Any help appreciated.