claws/BH1750 #37: Keep Returning 0 Value

briantstev, 2018-01-10:
Dear Friend,

I have an issue with your library. 
[image]

It keep returning 0 from sensor value. I copy and paste your code. Also already put your library file in place.

Any advice will be so helpful. 

Thank you

claws, 2018-01-11:
I have installed the library freshly into Arduino 1.8.5 then I cut and pasted the example from the README as you say you did. I am unable to replicate the problem you are seeing. I see:
```
BH1750 Test
Light: 23 lx
Light: 23 lx
Light: 23 lx
```

Today I updated the library to report more error information to the serial interface. This might help diagnose the problem you are having. Could you update to the latest version (i.e. remove the existing BH1750 library from {Arduino Path}/libraries, re-download it and re-install it) and then re-test?
 
Can you also provide any more information such as what platform (e.g. Win, OSX, Linux) you are running on, and exactly what code you are running?

briantstev, 2018-01-11:
It is very kind of you sir, please give me a moment to try it first, and let us see.

briantstev, 2018-01-13:
Hi mr. Claws.

Very sorry for late response. 
After further read on previous issues, I might think this is simply bad sensors.

So, I end up 
(1) buying some new other light sensors like you use GY30, 
(2) update my Arduino IDE to 1.8.5, previously 1.8.4. 
(3) test several boards.
(4) Copy and paste your example README.
(5) I use OSX environment 

Here are some photo that might help you to help me out :) 

p.s different photo, different light sensor.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

briantstev, 2018-01-13:
Mr. Claws,

Please give me more time to check on several boards and other type of light sensors like GY302

I'll update soon

briantstev, 2018-01-13:
Mr. Claws, 

Your update works like a charm!

It works, and I can't be more happier tonight :) 

Thank you, you are Light Sensor Library Legend! hehehe

[image]
[image]
[image]
[image]

claws, 2018-01-16:
Glad to hear it is working for you.

closed 2018-01-16 by claws

Guihgo, 2022-07-13:
Try begin module at mode ONE_TIME_HIGH_RES_MODE

```
if (!lightMeter.begin(BH1750::ONE_TIME_HIGH_RES_MODE))
    {
        Serial.println(F("Could not find a valid BH1750 sensor, check wiring!"));
        has_lightSensor = false;
    }
```