claws/BH1750 #43: Compile warning

pavanave, 2018-03-07:
Hi
first of all I'm sorry for my english! 
I have a ESP 32S board and the three examples in the your library don't work and give me a compile error:

"Compiling library "BH1750-master"
"/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/heap" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/spiffs" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/coap" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/console" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/expat" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/json" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/soc" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10900 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="esp32"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/cores/esp32" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/variants/esp32" "-I/Users/michele/Documents/Arduino/hardware/espressif/esp32/libraries/Wire/src" "-I/Users/michele/Documents/Arduino/libraries/BH1750-master" "/Users/michele/Documents/Arduino/libraries/BH1750-master/BH1750.cpp" -o "/var/folders/gl/m2zph6n55lsdyt5bys669rcr0000gn/T/arduino_build_28944/libraries/BH1750-master/BH1750.cpp.o"
/Users/michele/Documents/Arduino/libraries/BH1750-master/BH1750.cpp: In member function 'uint16_t BH1750::readLightLevel(bool)':
/Users/michele/Documents/Arduino/libraries/BH1750-master/BH1750.cpp:144:21: warning: overflow in implicit constant conversion [-Woverflow]
   uint16_t level(NAN);
                     ^
/Users/michele/Documents/Arduino/libraries/BH1750-master/BH1750.cpp:161:10: error: enumeration value 'UNCONFIGURED' not handled in switch [-Werror=switch]
   switch (BH1750_MODE) {
          ^
/Users/michele/Documents/Arduino/libraries/BH1750-master/BH1750.cpp:161:10: error: enumeration value 'CONTINUOUS_HIGH_RES_MODE' not handled in switch [-Werror=switch]
/Users/michele/Documents/Arduino/libraries/BH1750-master/BH1750.cpp:161:10: error: enumeration value 'CONTINUOUS_HIGH_RES_MODE_2' not handled in switch [-Werror=switch]
/Users/michele/Documents/Arduino/libraries/BH1750-master/BH1750.cpp:161:10: error: enumeration value 'CONTINUOUS_LOW_RES_MODE' not handled in switch [-Werror=switch]
cc1plus: some warnings being treated as errors
Using library Wire at version 1.0 in folder: /Users/michele/Documents/Arduino/hardware/espressif/esp32/libraries/Wire 
Using library BH1750-master at version 1.1.3 in folder: /Users/michele/Documents/Arduino/libraries/BH1750-master 
exit status 1
Error compiling for board ESP32 Dev Module."

I2C scanner give me the address 0x23.

I have tried the same sketch on an ESP8266 board and the sensor work great!

Please, can you help me?

coelner, 2018-03-07:
`cc1plus: some warnings being treated as errors` ignore warnings and don't treat them as error.
Or put a `default: break;` in line  174

pavanave, 2018-03-08:
Great!
I have change the line and Arduino IDE have compiled the sketch!
Resolved the problem
Thank you so much!

claws, 2018-09-23:
Seems resolved now.

closed 2018-09-23 by claws