claws/BH1750 #49: Huge difference in readings CONTINUOUS_HIGH_RES_MODE vs CONTINUOUS_HIGH_RES_MODE_2

wilhelmzeuschner, 2018-07-20:
I get very different readings, which differ by a factor of ~2:
1) `CONTINUOUS_HIGH_RES_MODE` gives me: 160 "lx" under a bright LED bulb.
2) `CONTINUOUS_HIGH_RES_MODE_2` gives me: 320 "lx" in the same environment.

Example:
1) 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 161 | 
2) 227 | 234 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | 324 | (at first I was covering the sensor a bit)

Which one of these readings is correct and what went wrong with the other one? Sadly I cannot really guess which one is correct because I cant really put them in perspective, i.e. I have no idea what to expect.

coelner, 2018-07-31:
@wilhelmzeuschner #46 it is currently in beta status, but your problem is solved 

https://github.com/coelner/BH1750/blob/a30562d2783f6bfafb75190a06fd950f9b58a68f/BH1750.cpp#L285-L287

claws, 2018-10-03:
Closing this issue as #46 is now merged.

closed 2018-10-03 by claws