claws/BH1750 #53: Usage information MTreg

coelner, 2018-10-24:
As you can see, I used the MTreg value to capture outdoor light. 
[image]

It reveals an error. The sensor is in deep sleep, therefore it restart completely and the saved MTreg value in the library get lost. But the sensor itself uses another MTreg value. I have no idea how to solve the problem in an nice way, but some thoughts:

1. include in `begin()` some resetting to the default MTreg
2. get/set method to manipulate the saved MTreg in the library space
3. inform the user to take care of this behavior
4. (find a hidden possibility to get the current MTreg from the sensor itself)

bjarnebuchmann, 2019-07-31:
I can confirm this issue.
The problem is that the BH1750 library does not always write the default MTreg to the device upon begin(). Rather, it assumes (sometimes falsely so) that the device uses the default MTreg on start. That is not true, if the program is reset (eg waking from deep sleep or rebooting) without giving a full power interrupt, which could reset MTreg on the BH1750.
The solution seems straight forward: setup() should always call

` setMTreg( (byte) BH1750_MTreg )`
or alt
` setMTreg( (byte)BH1750_DEFAULT_MTREG )`

to ensure that MTreg on device (BH1750) and object (library) mathches.

/Bjarne

bjarnebuchmann, 2019-07-31:
As a work-around, I will suggest to manually call `configure()` and `setMTreg()` right after `begin()`, as per:
```
setup(){
...
  lightMeter.begin(BH1750::CONTINUOUS_HIGH_RES_MODE);
  lightMeter.configure(BH1750::CONTINUOUS_HIGH_RES_MODE);
  lightMeter.setMTreg(BH1750_DEFAULT_MTREG);
...
}
```

coelner, 2020-08-14:
https://github.com/claws/BH1750/blob/38e9f378e07f0740f80a9a7700377ecdddcc7836/src/BH1750.cpp#L79
@claws what about this? 
`return (configure(mode) && setMTreg((byte) BH1750_DEFAULT_MTREG))`

commit 1c225367 (2021-01-03): MTReg fix #53
closed 2021-01-03 by claws