claws/BH1750 #55: add this library to Arduino Library Manager

iZhangHui, 2018-12-13:
Hi claws, could you add this library to Arduino Library Manager, then users can install and update this library easily. Thank you!

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

gilmaimon, 2019-01-21:
And may I add, after you do that, consider adding a badge from [ArduBadge](https://www.ardu-badge.com/) to notify users that your library is available on the Library Manager =)

bjarnebuchmann, 2019-07-31:
I would also love that this library went into the standard Arduino Library Manager. It is convenient not only for initial install, but also to signal that updates are available.
In any case, thanks for a good library.
/Bjarne

claws, 2019-08-01:
I'm happy to receive pull requests if you would like to propose a change that achieves this? With so many other things on the go it's hard for me to find the time.

Onno-Dirkzwager, 2020-05-10:
@claws are you still interested? I would like to help!

claws, 2020-05-10:
Yes, still interested.

Onno-Dirkzwager, 2020-05-10:
I hope to make a pull request this week. And I will send you some instructions for the final steps.

Onno-Dirkzwager, 2020-05-12:
Hi @claws,

**Description**
As you saw in the PR I changed some info to be more desciptive. If you search the current Library Manager you will find 3 other libraries. 

If people have to choose a library they tend to first try the more "complete" looking description first.

I read into all of them and as far as I can see yours stands out for supporting more archintectures. So I put this in the description. Feel free to change....

**Library name**
It's totaly up to you but if you have ever considerd changing the name, this is the time, once released as a library it can be a annoying process. If you do, make it stick out from the others that all have  more or less the same name.

**ArduBadge**
Added as @gilmaimon suggested. You might want to change the location...

**Github release**
Before informing the Arduino team about your library you will need to make an Github release. Following is based on personal experience and the [Arduino FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ)
1. click releases
2. click Draft a new release
3. Add your tag 1.1.4 (for now & future releases it's important that the tag reflects the version in your library files (.json & .properties) if it does not match Arduino will not accept it!)
4. Title & Description is up to you. They do not effect your library name! Check [other ](https://github.com/iotappstory/ESP-Library/releases) libraries for inspiration.
5. This is not a pre-release. Might come in handy for future updates. But from my personal experience the end user does not really understand.....
6. click Publish release

**Informing Arduino**
Open an issue on Arduino's GitHub, specifying the URL of the repository from where to download your library. If you have multiple libraries to submit you are welcome to do them all in a single issue.

Arduino had a job that runs every hour. If a new library has been released, you can expect it to be listed within the hour, but it might take longer.

Arduino "scans" your Github releases (tags) and if all is well adds them to the library manager. 

Ill help you with the first release! So if it fails let me know! Further I would suggest reading the [FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) as suggested by @iZhangHui 

Good luck!

Onno-Dirkzwager, 2020-06-02:
@claws 
I saw you did some things in preparation. And then it seemed to stop. All ok? Anything I can do?

claws, 2020-06-02:
Yeah, sorry. Just busy with work stuff and then exhausted. I'll get to it soon.

claws, 2020-06-07:
I have created the release and raised the PR to get the package added to the Arduino Library Manager list.

Onno-Dirkzwager, 2020-06-09:
Congratulations @claws,
your first library is now available in the Arduino Library Manager!
@iZhangHui @gilmaimon @bjarnebuchmann

claws, 2020-06-09:
@Onno-Dirkzwager Thanks for your help.

closed 2020-06-09 by claws