claws/BH1750 #56: Hello, the library dont really want to work with my eps32s, anyone?

marius240, 2019-03-14:
my code:
```#include <BH1750.h>
#include <Wire.h>

BH1750 lightMeter(0x5C);

void setup() {
  // put your setup code here, to run once:
  Wire.begin();
  Serial.begin(115200);
  lightMeter.begin(BH1750::CONTINUOUS_HIGH_RES_MODE);
  delay(2000);

}

void loop() {
  // put your main code here, to run repeatedly:

  uint16_t lux = lightMeter.readLightLevel();
  Serial.print("Light level: ");
  Serial.println(lux);

  delay(5000);

}

```
The cabling:

D21 -> SDA
D22 -> SCL
GND & ADDR -> gnd
VCC -> 3.3V

marius240, 2019-03-14:
btw my it gives me error: [BH1750] ERROR: received NACK on transmit of address
in the start and then for the rest of the time just:
Device is not configured!
Light level: 65534

marius240, 2019-03-14:
the board is NodeMCU ESP32S

marius240, 2019-03-14:
i2cdetect just gives me 
[image]

marius240, 2019-03-14:
Need help ASAP!
@claws @wridgers @goebish @NuTTeR  

thanks <3

goebish, 2019-03-14:
try with
Wire.begin(21, 22);

marius240, 2019-03-14:
@goebish nope aint working

marius240, 2019-03-14:
still same error, just ERROR: received NACK on transmit of address

goebish, 2019-03-14:
Have you tried to add pull up resistors between SDA<->3.3V and SCL<->3.3V ?
I'm not sure Wire.begin takes care of that.
If that doesn't work, then I don't know, have you tried with another i2c device such as a small oled display ?

marius240, 2019-03-14:
nope dont have another ic2 device, and dont think i got a pull up resister,

marius240, 2019-03-14:
but im pretty sure there is no need for a pull up resistor

goebish, 2019-03-14:
just use any 1k-30k resistors, that should be fine, you need 2

marius240, 2019-03-14:
so a resistor between D21 -> SDA and D22 -> SCL ?

marius240, 2019-03-14:
why should a resistor change anything? just asking :D

goebish, 2019-03-14:
no, between SDA -> 3.3V & SCL -> 3.3V

goebish, 2019-03-14:
because i2c lines need to be pulled up
https://rheingoldheavy.com/i2c-pull-resistors/

marius240, 2019-03-14:
not sure what you mean, there  is no between SDA and 3.3v i think
photo otw

goebish, 2019-03-14:
well, just connect a 10k or so resistor between +3.3V and SDA ... sma for SCL

marius240, 2019-03-14:
im trying to say, i connect direct from the board to scl, there is no connection to 3v between i can interact with

marius240, 2019-03-14:
[image]

marius240, 2019-03-14:
i have a 3v pin ofc, but i sounds weird if i should hook the up to them

marius240, 2019-03-14:
btw the blue line is ground and the red is 3.3v

marius240, 2019-03-14:
@goebish

goebish, 2019-03-14:
yes, just connect this 3.3V line to SDA through a resistor ... same for SCL

marius240, 2019-03-14:
lul ok,

goebish, 2019-03-14:
don't worry, you can't get the magic smoke out just by adding pull up resistors ;)

marius240, 2019-03-14:
hmmm it didnt work just hooking 3.3v onto scl and sda

marius240, 2019-03-14:
without resister tho

goebish, 2019-03-14:
... don't do that!
resistors are important!

marius240, 2019-03-14:
oh something changed, when i unplug both 3.3v it says ERORR: received NACK on transmit of address
but if i connect 3.3v to either one or both i get ERROR: received NACK on transmit of data

marius240, 2019-03-14:
@goebish

marius240, 2019-03-14:
get same result, when im using 0x23 and 0x5c
i thought it maybe could be cause that address was wrong

goebish, 2019-03-14:
At least that seems better with the pull up resistors, but I don't understand why you get an error on transmit of data then, the library is only using standard Arduino lib commands, nothing specific to a platform.
Have you tried the i2c scanner with pull up resistors ?
you can also try to use GPIO 16 & 17
Wire.begin(16, 17); // sda, scl
don't forget to change your wiring accordingly

marius240, 2019-03-14:
ok, i just tryed running i2cdetect, while have connected the 3.3v's, and it just give a billion errors per mili second, so i guess i need a resistor

goebish, 2019-03-14:
There's something I don't understand, are you connecting scl & sda directly to +3.3V ?
Never do that! Use resistors or nothing at all, but don't connect directly!

marius240, 2019-03-14:
just got the error with the address

goebish, 2019-03-14:
Use resistors between 1k and 30kΩ, not 150kΩ or 1.5MΩ, are you reading my posts ?

marius240, 2019-03-14:
yep

marius240, 2019-03-14:
oh sorry did just catch the 1k - 30k / i forgot it

coelner, 2019-03-17:
Did you solder the pins to the boards or do you just push it through? Solder it and check the resistance with a multimeter.  If you solder the pins as seen in the picture then it is wrong. (The board itself contains usually the needed pullup resistors)
 - Do you have another board with an esp8266?
 - Do you have another i2c sensor?

marius240, 2019-03-17:
Yea the pins is soldered, but the board is a esp32, I have a esp8266, but I need it to work on esp32

coelner, 2019-03-17:
I'm aware of your board, but I want to know whether the sensor works with another board.

coelner, 2019-03-17:
I'm aware of your board, but I want to know whether the sensor works with another board.
The library supports the esp32, I tested it right away and it works flawless.

marius240, 2019-03-17:
hmm, which address are you using?

coelner, 2019-03-17:
The default one (0x23). I use the lolin32lite, Pin 22 and 21 for i2c.

marius240, 2019-03-17:
ye i use pin 22 and pin 21 to,  have tryed both 0x23 and 0x5C

coelner, 2019-03-17:
To sum it up. Your wiring, your board, your sensor. Not the library.

marius240, 2019-03-18:
hm ok

marius240, 2019-03-18:
so something is wrong with the wirering?

coelner, 2019-03-18:
I think so. 
 * The Pins are soldered with the wrong side to the board. This could be a connection failure
 * These breadboards are sometimes pretty lousy and have a high resistance or have a shortcut
 * Is power at the power rail on the breadboard?
 * which name has your board? GY-302?
 * The I2C address should be 0x23, because you tied ADDR to GND. (you don't need it)
 * which SDK verson do you use?

marius240, 2019-03-21:
i use the latest arduinoIDE, im not sure what the power rail is, what do you mean with soldered to the wrong side?

coelner, 2019-03-21:
- The power rail is on the board the parts with the blue and red line. You need to supply power. Sometimes those power rails are splitted. Take a multimeter and measure the voltage at the green wires. 
 - The soldered pins with the black plastic bottom are wrong. they should have been put through up side down. Measure the voltage here too, it should be 3.3V.
- Did you check the Board Manager in the arduino IDE for updates?

IoTThinks, 2020-02-29:
You may need to pull up the SDA and SCL before Wire.begin(SDA, SCL);
pinMode(SDA, INPUT_PULLUP);
pinMode(SCL, INPUT_PULLUP);

And make sure the BH1750 is...physically connected.