claws/BH1750 #57: Lower bits of MTreg

kenjif70, 2019-07-15:
https://github.com/claws/BH1750/blob/a49b08550a8f46d3c7672a3774103cd01dc8186b/BH1750.cpp#L156

Shouldn't we keep 5 bits rather than 3 bits?

claws, 2019-07-29:
I think you are correct. Do you want to submit a pull request?

kenjif70, 2019-07-29:
Thanks for the comment. Yup, I can do that if you prefer. But I've not set up the repo/branch yet, so I'm perfectly fine/happy if you could make the change on behalf of me.

claws, 2019-07-29:
Do you have a use case that can test the change? If so can you test the change in the merge request (#58) branch I just raised to applied the fix?

kenjif70, 2019-07-29:
Cool! I just confirmed that, with the fix, readLightLevel gives a consistent value regardless of the sensitivity. Before the fix, MTreg=39 and MTreg=63 give very different numbers, for example.

By the way, another thing I realized while testing is that readLightLevel returns an inconsistent number if it's called right after calling setMTreg with the mode = ONE_SHOT_*. setMTreg has _delay_ms when the mode is CONTINUOUS_* but not ONE_SHOT_*, and this inconsistent behavior doesn't happen in CONTINUOUS_* modes.

claws, 2019-07-29:
Thanks for catching this issue. I've merged #58 which provides the fix and raised another issue (#59) to capture the last issue you observed.

closed 2019-07-29 by claws