claws/BH1750 #61: Frequent calls to readLightLevel prevents measurement update

bjarnebuchmann, 2019-07-31:
It seems that in CONTINUOUS modes, frequent calls to readLightLevel prevents the bh1750 module to actually update the measurement. Presumably, the measurement is interrupted by the i2c communication and has to be restarted. 
Thus, if I (say) set 
`lightMeter.configure(BH1750::CONTINUOUS_HIGH_RES_MODE);`
and then call readLightLevel with only 100 (or even 120) millis delays, then I keep getting the same value back.
If I increase to, say, 180 millis (max time to make measurement) ,then I get an updated value every time.

If there already is a caveat about this usage, then I apologize for reporting.

I do not see an easy way to circumvent this, except storing (in the object) the previous measurement and time-of-measurement, and not then checking time vs previous time-of-measurement to see if an updated measurement should be ready.

Obviosly, I can avoid the problem by not calling `readLightLevel()` often. I was just surprised by the effect, and want to report it. 

Best,

/Bjarne

coelner, 2019-08-14:
No, that looks like a bug. 
https://github.com/claws/BH1750/blob/06436316cad24ae375e92e62e2cc61a6fea83584/BH1750.cpp#L220-L224
We should check for continuous mode and do not reconfigure the sensor.

coelner, 2019-08-14:
Can you replace it accordingly with the following and test it?

```
float BH1750::readLightLevel(bool maxWait) {

  if (BH1750_MODE == UNCONFIGURED) {
    Serial.println(F("[BH1750] Device is not configured!"));
    return -2.0;
  }

  // Measurement result will be stored here
  float level = -1.0;

  // Wait for measurement to be taken.
  // Measurements have a maximum measurement time and a typical measurement
  // time. The maxWait argument determines which measurement wait time is
  // used when a one-time mode is being used. The typical (shorter)
  // measurement time is used by default and if maxWait is set to True then
  // the maximum measurement time will be used. See data sheet pages 2, 5
  // and 7 for more details.
  // A continuous mode measurement can be read immediately after re-sending
  // the mode command.

  switch (BH1750_MODE) {

    case BH1750::ONE_TIME_LOW_RES_MODE:
        // Send mode to sensor
  Wire.beginTransmission(BH1750_I2CADDR);
  __wire_write((uint8_t)BH1750_MODE);
  Wire.endTransmission();
      maxWait ? _delay_ms(24 * BH1750_MTreg/(byte)BH1750_DEFAULT_MTREG) : _delay_ms(16 * BH1750_MTreg/(byte)BH1750_DEFAULT_MTREG);
      break;
    case BH1750::ONE_TIME_HIGH_RES_MODE:
    case BH1750::ONE_TIME_HIGH_RES_MODE_2:
      // Send mode to sensor
  Wire.beginTransmission(BH1750_I2CADDR);
  __wire_write((uint8_t)BH1750_MODE);
  Wire.endTransmission();
      maxWait ? _delay_ms(180 * BH1750_MTreg/(byte)BH1750_DEFAULT_MTREG) :_delay_ms(120 * BH1750_MTreg/(byte)BH1750_DEFAULT_MTREG);
      break;
    default:
      break;
  }

  // Read two bytes from the sensor, which are low and high parts of the sensor

```