claws/BH1750 #62: [BH1750] Device is not configured! ESP8266 afterESP.deepSleep

ahostn, 2019-10-04:
`/*

  Advanced BH1750 library usage example

  This example has some comments about advanced usage features.

  Connection:

    VCC -> 3V3 or 5V
    GND -> GND
    SCL -> SCL (A5 on Arduino Uno, Leonardo, etc or 21 on Mega and Due, on esp8266 free selectable)
    SDA -> SDA (A4 on Arduino Uno, Leonardo, etc or 20 on Mega and Due, on esp8266 free selectable)
    ADD -> (not connected) or GND

  ADD pin is used to set sensor I2C address. If it has voltage greater or equal to
  0.7VCC voltage (e.g. you've connected it to VCC) the sensor address will be
  0x5C. In other case (if ADD voltage less than 0.7 * VCC) the sensor address will
  be 0x23 (by default).

*/

#include <Wire.h>
#include <BH1750.h>
#define durationSleep 10
/*
  BH1750 can be physically configured to use two I2C addresses:
    - 0x23 (most common) (if ADD pin had < 0.7VCC voltage)
    - 0x5C (if ADD pin had > 0.7VCC voltage)

  Library uses 0x23 address as default, but you can define any other address.
  If you had troubles with default value - try to change it to 0x5C.

*/
BH1750 lightMeter(0x23);

void setup() {

  Serial.begin(115200);

  // Initialize the I2C bus (BH1750 library doesn't do this automatically)
  Wire.begin(D7, D8);
  // On esp8266 you can select SCL and SDA pins using Wire.begin(D4, D3);

  /*

    BH1750 has six different measurement modes. They are divided in two groups;
    continuous and one-time measurements. In continuous mode, sensor continuously
    measures lightness value. In one-time mode the sensor makes only one
    measurement and then goes into Power Down mode.

    Each mode, has three different precisions:

      - Low Resolution Mode - (4 lx precision, 16ms measurement time)
      - High Resolution Mode - (1 lx precision, 120ms measurement time)
      - High Resolution Mode 2 - (0.5 lx precision, 120ms measurement time)

    By default, the library uses Continuous High Resolution Mode, but you can
    set any other mode, by passing it to BH1750.begin() or BH1750.configure()
    functions.

    [!] Remember, if you use One-Time mode, your sensor will go to Power Down
    mode each time, when it completes a measurement and you've read it.

    Full mode list:

      BH1750_CONTINUOUS_LOW_RES_MODE
      BH1750_CONTINUOUS_HIGH_RES_MODE (default)
      BH1750_CONTINUOUS_HIGH_RES_MODE_2

      BH1750_ONE_TIME_LOW_RES_MODE
      BH1750_ONE_TIME_HIGH_RES_MODE
      BH1750_ONE_TIME_HIGH_RES_MODE_2

  */

  // begin returns a boolean that can be used to detect setup problems.
  if (lightMeter.begin(BH1750::CONTINUOUS_HIGH_RES_MODE)) {
    Serial.println(F("BH1750 Advanced begin"));
  }
  else {
    Serial.println(F("Error initialising BH1750"));
  }

  float lux = lightMeter.readLightLevel();
  Serial.print("Light: ");
  Serial.print(lux);
  Serial.println(" lx");
  delay(1000);

  ESP.deepSleep(durationSleep * 1000000);
}

void loop() {

}`

Any ideas why deepSleep breaks the library ?

claws, 2019-10-05:
1. The formatting of the code in the issue report makes it a little hard for me to read. However, it looks like you have moved the code that typically resides in the ``loop`` function up into the ``setup`` function, leaving the ``loop`` function empty. Was this intentional?
2. Have you confirmed that your devices works fine with one of the example scripts (i.e. without the ESP deepSleep addition)?
3. Does the problem only appear once you add the deepSleep line?

ahostn, 2019-10-07:
Hello. I'm pretty sure now that I have a hardware problem, your code is good.
If I connect the GY-30 sensor, ESP8266 doesn't even connect to WiFi even if I use external power for the sensor. If I attach the sensor (+3.3V line) after wifi is connected, it works fine... until reset/sleep. i2c bus is not stuck, so it must be something with the sensor.
Even your example code now works only if I connect power to the sensor after booting up, but it worked for the first time when I tried it.. :)

closed 2019-10-10 by claws

pahanadithya, 2023-12-16:
#include <BH1750.h>
#include <Wire.h>

BH1750 lightMeter;

void setup() {
  Serial.begin(9600);

  Wire.begin();

  lightMeter.begin();

  Serial.println(F("BH1750 Test begin"));
}

void loop() {
  float lux = lightMeter.readLightLevel();
  Serial.print("Light: ");
  Serial.print(lux);
  Serial.println(" lx");
  delay(1000);
}

This is my code, I always get an error as "[BH1750] Device is not configured!
Light: -2.00 lx" , how can i fixed that error. I can't ger proper output. 
[image]

ahostn, 2023-12-17:
If you’re using deepSleep, you should move all your code in setup block, not main loop.