claws/BH1750 #64: Trouble using 2 sensors

Coelhomatias, 2020-04-17:
Hi, using this code:

`
#include<Arduino.h>
#include <Wire.h>
#include <BH1750.h>
BH1750 lightMeter1(0x23);
BH1750 lightMeter2(0x5C);
void setup(){
  Serial.begin(9600);
  Wire.begin(D1, D2);
  lightMeter1.begin(BH1750::CONTINUOUS_HIGH_RES_MODE_2);
  lightMeter2.begin(BH1750::CONTINUOUS_HIGH_RES_MODE_2);
  Serial.println(F("BH1750 Test"));
}
void loop() {
  uint16_t lux1 = lightMeter1.readLightLevel();
  uint16_t lux2 = lightMeter2.readLightLevel();
  Serial.print("Light1: ");
  Serial.print(lux1);
  Serial.print(" lx. Light2: ");
  Serial.print(lux2);
  Serial.println(" lx.");
  delay(1000);
}`

in a nodeMCU, like sugested in #25 but the output is the same for both, only reads from 0x23

Coelhomatias, 2020-04-17:
Device finds both sensors

[image]

Coelhomatias, 2020-04-17:
Found the issue!! the begin() function also has a default address.
Problem corrected with this:

```C++
#include<Arduino.h>
#include <Wire.h>
#include <BH1750.h>
BH1750 lightMeter1(0x23);
BH1750 lightMeter2(0x5C);
void setup(){
  Serial.begin(9600);
  Wire.begin(D1, D2);
  lightMeter1.begin(BH1750::CONTINUOUS_HIGH_RES_MODE_2, 0x23);
  lightMeter2.begin(BH1750::CONTINUOUS_HIGH_RES_MODE_2, 0x5C);
  Serial.println(F("BH1750 Test"));
}
void loop() {
  uint16_t lux1 = lightMeter1.readLightLevel();
  uint16_t lux2 = lightMeter2.readLightLevel();
  Serial.print("Light1: ");
  Serial.print(lux1);
  Serial.print(" lx. Light2: ");
  Serial.print(lux2);
  Serial.println(" lx.");
  delay(1000);
}
```
@claws you should fix this. Closing the issue

closed 2020-04-17 by Coelhomatias

claws, 2020-04-18:
For future reference, this use case is demonstrated by the [example](https://github.com/claws/BH1750/blob/master/examples/BH1750two_i2c/BH1750two_i2c.ino) showing how different I2C and Wire instances can be used.
This particular use case could potentially be made clearer by adding something to the README.