claws/BH1750 #71: [TIP] Device not configured - temporary fix.

Zontex, 2020-10-23:
Hi All,
So I've seen people having this issue "device not configured" for me, I know it's not hardware issue because I've been running MicroPython without a problem and the BH1750 sensor is soldered to a custom PCB I've made.

The address in my case is 0x5C I've tried to change it "softly" it didn't work, changed the source files didn't work till I found a fix by looking into one of the advanced examples.

After changing lightMeter.begin() with the following:
`lightMeter.begin(BH1750::CONTINUOUS_HIGH_RES_MODE, 0x5C, &Wire);`

It fixed my problem, the sensor is working perfectly fine now. I have no idea why as I'm a Python not C or C++ Programmer but would love the author @claws to take a look over it and see if can find the issue.

I've spent about 25 minutes trying to re-solve the problem till I came with this solution, hope it can save some time for others as well that having the same issue as me.

Complete code:
`
#include <Wire.h>

/*

  Example of BH1750 library usage.

  This example initialises the BH1750 object using the default high resolution
  continuous mode and then makes a light level reading every second.

  Connection:

    VCC -> 3V3 or 5V
    GND -> GND
    SCL -> SCL (A5 on Arduino Uno, Leonardo, etc or 21 on Mega and Due, on esp8266 free selectable)
    SDA -> SDA (A4 on Arduino Uno, Leonardo, etc or 20 on Mega and Due, on esp8266 free selectable)
    ADD -> (not connected) or GND

  ADD pin is used to set sensor I2C address. If it has voltage greater or equal to
  0.7VCC voltage (e.g. you've connected it to VCC) the sensor address will be
  0x5C. In other case (if ADD voltage less than 0.7 * VCC) the sensor address will
  be 0x23 (by default).

*/

#include <Wire.h>
#include <BH1750.h>

BH1750 lightMeter(0x5C);

void setup(){
  // start serial communication
  Serial.begin(115200);
  // start I2C on pins 4 and 15
  Wire.begin(4,15);
  // initialize the bh1750
  lightMeter.begin(BH1750::CONTINUOUS_HIGH_RES_MODE, 0x5C, &Wire);
}

void loop() {

  float lux = lightMeter.readLightLevel();
  Serial.print("Light: ");
  Serial.print(lux);
  Serial.println(" lx");
  delay(3000);

}
`

claws, 2020-10-23:
Thanks for the feedback, I'll look into it when I get some time.

coelner, 2020-10-28:
@claws I don't see anything special here. we could test in the configure() method whether the chip respond to 0x50 address or not. But this does not have an ID register to identify this chip.

coelner, 2020-11-01:
[ ] How does the sensor respond to a changed ADDR line while in operation? 

If the sensor accepts a changed address we can keep it in the `begin()`, if it doesn't we should use the address only at declaration time.
But in general I assume that a sensor is fixed to a specific address through the pcb. I see no benefit from changing the address via gpio, because in most cases this sensor is attached to a specific I2C bus. There shouldn't be a problem with a sudden address conflict, should it? @claws

coelner, 2021-01-05:
The sensor allows an address change while running.