claws/BH1750 #77: BH1750 does not work on ESP32

sergioarjona5, 2021-04-28:
Hi guys,

First of all sorry for my English.  I have a problem with the BH1750, it tells me that it is not well in fact with the i2cdetec, it does not detect it and I no longer know what to do. I have tried everything, pullup resistors as well. I have also tried different BH1750 sensors. I have tested de pins and work. The only thing that I 

Here is my code:
#include <Wire.h>
#include <BH1750.h>
#include <i2cdetect.h>

BH1750 lightMeter;

void setup() {

  Serial.begin(115200);

  // Initialize the I2C bus (BH1750 library doesn't do this automatically)
  Wire.begin();
  // On esp8266 you can select SCL and SDA pins using Wire.begin(D4, D3);
  i2cdetect();
  if (lightMeter.begin()) {
    Serial.println(F("BH1750 Test begin"));
  }
  else {
    Serial.println(F("BH1750 Initialization FAILED"));
  }

}

void loop() {

  float lux = lightMeter.readLightLevel();
  Serial.print("Light: ");
  Serial.print(lux);
  Serial.println(" lx");
  
  delay(1000);

}

Here I show de wiring.
[image]

Thanks.

lefty01, 2022-08-07:
is this still an issue? works for me (now) with esp32.

ryancasler, 2022-11-01:
With the ESP32, depending on what board you are using the SDA and SCL default pins might be different.   If that's an ESP32 WROOM dev kit, then your SDA pin is 21 and your SCL pin is 22 by default.  Or you can simply use:
Wire.setPins(<SDA PIN>, <SCL PIN>);
and substitute whatever pin numbers you are using for SDA and SCL.  You just have to do this before the Wire.begin() command.

petobe, 2023-11-08:
@lefty01 Yep, for me its still an issue....but i just started with ESP32 :) and cant get it to work. I have an D1 Mini ESP32 from AZ-Delivery. Do you want t share how you got it to work?